MicrosoftDocs/windows-driver-docs #1404: What does a "NotSet" Submission Status represent?

GabeSJones, 2019-03-13:
It would be nice to have an explanation of Submission Status.  Particularly, I see "NotSet" in the field for drivers uploaded via the dashboard API, even after the signing process has been completed.  What does "NotSet" mean?  I never see it when uploading drivers via the web dashboard.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 5cd3829b-c6c5-2ee2-5741-86714cff2345
* Version Independent ID: ad072ec7-1388-0499-0e88-ea8074abb8cf
* Content: [Find and manage your hardware submission - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/find-hardware-submission)
* Content Source: [windows-driver-docs-pr/dashboard/find-hardware-submission.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/dashboard/find-hardware-submission.md)
* Product: **windows-hardware**
* GitHub Login: @dimanjar
* Microsoft Alias: **dimanjar**

DOMARS, 2019-03-18:
@EliotSeattle - Looks like dashboard, so assigning to you. 
The bot may need to be updated to assign dashboard.

EliotSeattle, 2019-03-21:
Thanks, @GabeSJones 
Working on getting accurate descriptions for you.

EliotSeattle, 2020-10-22:
Hi @GabeSJones 

Sorry that this sat. 

Yes, I believe the table is correct. For specific questions about your particular submission, Hardware Dev Center Support can investigate. To contact support and open a Hardware Dev Center support case for investigation, **first**, sign into your hardware dev center account, and **then** go to [https://docs.microsoft.com/windows-hardware/drivers/dashboard/hardware-dashboard-faq](https://docs.microsoft.com/windows-hardware/drivers/dashboard/hardware-dashboard-faq).

Find: 
 
**How do I contact Partner Center Support?**
If you are having problems accessing the Dashboard or need Dashboard Support please open a support ticket here: [https://developer.microsoft.com/windows/support](https://developer.microsoft.com/windows/support). 

Once there, select **Contact us**, **Dashboard issue**, and then **Hardware submissions & signing (all OS version)** from the dropdown menu.

Hopes this helps with future issues.
Eliot

tedhudek, 2020-11-03:
Believe NotSet is not a valid value and should be removed from the doc page but need to confirm

tedhudek, 2020-11-16:
Thanks for the query, @GabeSJones, and apologies for the delay.

I actually don't see `NotSet` in the code anymore, so maybe it was removed.   I've removed it from the doc page - changes should be live by the end of day.

closed 2020-11-16 by tedhudek