MicrosoftDocs/windows-driver-docs #1412: How can I use pykd in windbg preview？

tchklmxx, 2019-03-16:
I can copy  pykd.dll to Debuggers\x86\winext  directory to use pykd in older version
But how to do it now？to type 
.load c:\xxx\xxx\xxx\xxx\pykd.dll 
every time is not a good idea......

---
#### 文档详细信息

⚠ *请勿编辑此部分。 docs.microsoft.com ➟ GitHub 问题链接必须具有此部分。*

* ID: 8353dbe7-846a-f902-0f6c-f7e75ad846d8
* Version Independent ID: 6ec7d01f-69f9-5f7b-582b-0d63d9c40a06
* Content: [WinDbg Preview - Command line startup options - Windows drivers](https://docs.microsoft.com/zh-cn/windows-hardware/drivers/debugger/windbg-command-line-preview#feedback)
* Content Source: [windows-driver-docs-pr/debugger/windbg-command-line-preview.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/windbg-command-line-preview.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2019-03-28:
@aluhrs13 - Do we provide the ablity to load extension DLLs in Windbg Preview in an apps directory? Let me know if there is a Debuggers\x86\winext directory for Windbg Preview that I should document.

aluhrs13, 2019-03-29:
See https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/loading-debugger-extension-dlls

The options are:

1. Set _NT_DEBUGGER_EXTENSION_PATH to include the folder that contains your extension.
2. Run ".extpath" to see the default extension path and copy your extension to any of those locations.
3. Fully qualify the path of the binary when running ".load".

@DOMARS to answer your exact question, there's no exact folder path to document.

DOMARS, 2019-03-29:
@tchklmxx - See the three options listed above. 
As another venue for questions like this, [Windows Hardware WDK and Driver Development](https://social.msdn.microsoft.com/Forums/windowsdesktop/en-US/home?forum=wdk) can be a good resource.

closed 2019-03-29 by DOMARS

justanotheranonymoususer, 2020-01-26:
> * Set _NT_DEBUGGER_EXTENSION_PATH to include the folder that contains your extension.

Why not add it to the UI under "Debugging settings"? Like there's an option for source folders paths.

> * Run ".extpath" to see the default extension path and copy your extension to any of those locations.

One of the paths I see is:
`C:\Users\user\AppData\Local\Dbg\EngineExtensions`
even thought this folder doesn't exist. So looks like it answers the question, one can create and use it. Is it not by design? I assume it's worth documenting.