MicrosoftDocs/windows-driver-docs #1415: great  ariticle

RAJU2529, 2019-03-16:
i want to add  another example  PCl hardware ids. Thanks  to the authors for  article

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: e29e581e-02a0-dab0-83db-25ae1ca1aebb
* Version Independent ID: 21141809-e821-8e58-dcf2-be1d1bb9be87
* Content: [Step 1 The New Device is Identified - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/step-1--the-new-device-is-identified)
* Content Source: [windows-driver-docs-pr/install/step-1--the-new-device-is-identified.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/step-1--the-new-device-is-identified.md)
* Product: **windows-hardware**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

RAJU2529, 2019-03-16:
For example, thePCI graphics  card Hardware id ( INTEL )
1. PCI\VEN_8086&DEV_5916&SUBSYS_1A001043&REV_02

VID_8086 is the identifier of the vendor intel
DEV_5916 is the intel HD graphics  620
REV_02 is the revision identifier of the device.

For example, thePCI graphics  card Hardware id ( NVIDIA )
2
[image]
[image]
[image]

. PCI\VEN_10DE&DEV_134D&SUBSYS_142E1043&REV_A2

VID_10DE is the identifier of the vendor nvidia
DEV_134D is the Nvidia Geforce 940mx
REV_A2 is the revision identifier of the device.

tedhudek, 2019-06-18:
Thanks @RAJU2529 for the feedback.  I added some info in a4057b3 with other examples.

closed 2019-06-18 by tedhudek