MicrosoftDocs/windows-driver-docs #1425: WPP_CLEANUP的问题

kouzhudong, 2019-03-26:
为何在DriverUnload里调用了WPP_CLEANUP，还是会出现：
DRIVER_VERIFIER_DETECTED_VIOLATION (c4)
A device driver attempting to corrupt the system has been caught.  This is
because the driver was specified in the registry as being suspect (by the
administrator) and the kernel has enabled substantial checking of this driver.
If the driver attempts to corrupt the system, bugchecks 0xC4, 0xC1 and 0xA will
be among the most commonly seen crashes.
Arguments:
Arg1: 00000000000000dd, Unloading driver that forgot to call EtwUnregister.
Arg2: fffff880046a76d0, Address where the culprit driver called EtwRegister.
Arg3: fffff88004665000, Start address of the culprit driver.
Arg4: 0000000000000000, Address of the leaked ETW_REG_ENTRY structure.
我怀疑WPP_CLEANUP的代码有问题，或者Windows的机制有问题，如：verifier。

EliotSeattle, 2019-06-18:
Hello @kouzhudong 
What page are you referencing here? I'm don't understand what you're saying without knowing the page.

Thanks,
@EliotSeattle

kouzhudong, 2019-06-19:
应该是提交给windbg的帮助文档。

kouzhudong, 2019-06-19:
https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0xc4--driver-verifier-detected-violation

应该提交到这里。

EliotSeattle, 2019-06-19:
I see that this was opened as #1623 in that topic, so I am going to close this as a duplicate Issue.

closed 2019-06-19 by EliotSeattle