MicrosoftDocs/windows-driver-docs #1428: How and where to log from ELAM module?

CodeAbyss, 2019-03-27:
I want to log some information from the ELAM at run-time. The easiest way I could think is logging into vendor's name subkey under `\\Registry\\Machine\\ELAM\\VendorName`. 

The issue is when I do `ZwSetValueKey`, it returns `NT_SUCCESS `& `ZwQueryValueKey `too finds the `Value `successfully. Now when I load the ELAM hive from user mode, it don't contain my `Value `written to it. Kindly explain the behavior.

Also, where should we log data if required from the ELAM at run-time?

CodeAbyss, 2019-04-01:
No comments?

tedhudek, 2019-06-19:
Thanks for the feedback @CodeAbyss and apologies for the delay getting back to you.  We recommend using standard ETW logging.  Changes to the ELAM registry hive are not written to disk, so don't write log info there.  I've updated the page in question with a note containing this info.

The changes should appear on the live site within half an hour.

closed 2019-06-19 by tedhudek

CodeAbyss, 2019-06-20:
Thanks for the update.