MicrosoftDocs/windows-driver-docs #1430: Sentence is repeated

raghavmishra, 2019-03-27:
In section firewall and debuggers first sentence should be corrected as it contains repeated line.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: f5abd01e-3c7e-11a8-a6c6-62bfba0ef8b6
* Version Independent ID: dceb81b3-b20f-91d1-85d0-fa68c762482b
* Content: [Debug Universal Drivers - Step-by-Step Lab (Echo Kernel Mode) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/debug-universal-drivers---step-by-step-lab--echo-kernel-mode-#feedback)
* Content Source: [windows-driver-docs-pr/debugger/debug-universal-drivers---step-by-step-lab--echo-kernel-mode-.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/debug-universal-drivers---step-by-step-lab--echo-kernel-mode-.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2019-03-28:
@raghavmishra - Thanks for pointing this out - fix is on the way.

DOMARS, 2019-05-02:
@raghavmishra - The fix to remove the duplicate sentence is live. Thanks for pointing out this issue.

closed 2019-05-02 by DOMARS