MicrosoftDocs/windows-driver-docs #1434: feedback

broughammer, 2019-03-29:
nice

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: cf044688-70f6-fb35-b787-dfc2eff1f256
* Version Independent ID: be66e130-a586-1270-6c9b-6a1fbd3e1563
* Content: [Overview of Storage Virtual Miniport Drivers - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/storage/overview-of-storage-virtual-miniport-drivers#feedback)
* Content Source: [windows-driver-docs-pr/storage/overview-of-storage-virtual-miniport-drivers.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/storage/overview-of-storage-virtual-miniport-drivers.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

EliotSeattle, 2019-03-29:
Thanks, @broughammer!

What was there in particular that was helpful to you? 

Eliot

lorihollasch, 2019-04-26:
@broughammer, thanks for your feedback. :)

closed 2019-04-26 by lorihollasch