MicrosoftDocs/windows-driver-docs #1438: Is "update a shipping label" API working now?

hatasum, 2019-04-02:
I am not sure if I can ask the following question here:
https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/update-a-shipping-label

Is "update a shipping label" API working now?
I try the api, it always return StatusCode: 422

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 68b9e39f-a269-82bb-d44a-162ed96fab57
* Version Independent ID: 03adbc66-9e14-11b0-407b-a2d4b09ee7b5
* Content: [Update a shipping label - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/update-a-shipping-label)
* Content Source: [windows-driver-docs-pr/dashboard/update-a-shipping-label.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/dashboard/update-a-shipping-label.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @balapv
* Microsoft Alias: **balapv**

hatasum, 2019-04-02:
In JSON request body,  not "targetingInfo", correct is "Targeting".

EliotSeattle, 2019-04-02:
Thank you, @hatasum.

Yes, the API should be working now.

Error 422 - Unprocessable Entity means that there are Validation failures. See [Error Codes](https://docs.microsoft.com/windows-hardware/drivers/dashboard/get-product-data#error-codes) in **Get product data**.

If you have issues with your specific submission, you can reach out for support.
1. **First**, sign into your hardware dev center account, and then 
2. Open a Hardware Dev Center support case for investigation. 
3. Go to 
[https://docs.microsoft.com/windows-hardware/drivers/dashboard/hardware-dashboard-faq](https://docs.microsoft.com/windows-hardware/drivers/dashboard/hardware-dashboard-faq)
and look for: 

**How do I contact Partner Center Support?**
If you are having problems accessing the Dashboard or need Dashboard Support please open a support ticket here: [https://developer.microsoft.com/windows/support](https://developer.microsoft.com/windows/support). Select **Contact us**, **Dashboard issue**, and then **Hardware submissions & signing (all OS version)** from the dropdown menu.

I will look into "targetingInfo" v. "Targeting" in the JSON request body and will make any necessary corrections.

Thanks,
Eliot

EliotSeattle, 2019-04-02:
And in the source, I see that  "targetingInfo" is correct, as are the values associated with it: "chids", "hardwareIds", "restrictedToAudiences", and "businessJustification".
Thanks!

closed 2019-04-02 by EliotSeattle

hatasum, 2019-04-03:
http://download.microsoft.com/download/C/F/4/CF404E53-87A0-4204-BA13-A64B09A237C1/HardwareApiCSharpSample.zip
the sample's mark says:use Targeting instead of TargetingInfo
Indead, "Targeting" is working...
whatever, thanks your reply.