MicrosoftDocs/windows-driver-docs #1443: Older WinDbg symbols available?

ladislav-zezula, 2019-04-03:
Is there a way to download older symbols and original binaries from MS symbol server?
Like the ones from Windows XP?

Thanks in advance.

**More info**
It seems that some time ago (not exactly sure when), Microsoft decided to cleanup the symbol server store. Since then, older symbols seem to be no more there and the files have changed names, because they are no longer compressed (or they seem to be, at least).

**===== OLD STRUCTURE ====**
Files were stored compressed, as CAB files.
Example: calc.exe 5.1.2600.0 from vanilla Windows XP
http://msdl.microsoft.com/download/symbols/calc.exe/3B7D84101F000/calc.ex_
That file is no longer available. The same applies to e.g. kernel32.pdb (.pd_, respectively)

**===== NEW STRUCTURE ====**
Files are stored in plain (uncompressed) form, be it EXE or PDB.
Example: calc.exe v 10.0.0.17763.1 (x64)
Remote path is: http://msdl.microsoft.com/download/symbols/calc.exe/8F598A9EB000/calc.exe
Can download OK, e.g. with wget.

rodwiddowson, 2019-04-03:
+1 for @ladislav-zezula 's question.  

IMHO: The documentation [here](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/microsoft-public-symbols.md) would be the place to put the information....

DOMARS, 2019-04-03:
@ladislav-zezula - I'm not sure what we have for older Windows XP symbols on the symbol server and what format they are provided on the symbols server. Perhaps @aluhrs13 can comment on that.

aluhrs13, 2019-04-03:
> The Microsoft Symbol Server provides compressed versions of the symbol files. The files have an underscore at the end of the filename’s extension to indicate that they are compressed. For example, the PDB for ntdll.dll is available as ntdll.pd_. When SymProxy downloads a compressed file, it will store the file decompressed in the local file system. The DontUncompress registry key can be set to disable this behavior in SymProxy.

This portion of that page should be updated. The symbol server was migrated to different back-end that uses HTTP compression instead of CAB compression. This transition should be seamless for any debugger since there's logic in symsrv that check for both compressed and uncompressed symbols. 

For @ladislav-zezula 's original question, `http://msdl.microsoft.com/download/symbols/calc.exe/3B7D84101F000/calc.ex_` no longer exists but `http://msdl.microsoft.com/download/symbols/calc.exe/3B7D84101F000/calc.exe` still resolves fine and is the uncompressed version of the same file.

ladislav-zezula, 2019-04-04:
Thank you for the response.

closed 2019-04-04 by ladislav-zezula