MicrosoftDocs/windows-driver-docs #1445: OS Descriptors Specification Links

justinmcbride, 2019-04-04:
The links to the descriptor specifications lead to legacy pages that don't allow downloading of the actual specifications.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: e53fc82f-4f50-7564-a5dd-c1ad8df5cd77
* Version Independent ID: f21decfc-caec-36cf-d7f7-186f7a244e28
* Content: [Microsoft OS Descriptors for USB Devices - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/usbcon/microsoft-defined-usb-descriptors#feedback)
* Content Source: [windows-driver-docs-pr/usbcon/microsoft-defined-usb-descriptors.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/usbcon/microsoft-defined-usb-descriptors.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @duncanmacmichael
* Microsoft Alias: **dumacmic**

duncanmacmichael, 2019-04-04:
Hi @justinmcbride, thanks for finding this. Some brief searching around didn't turn up anything as to where those downloads went, so let me look further and see if I can track them down.

justinmcbride, 2019-05-10:
Hey, using archive.org, we found the document link to be as follows: download.microsoft.com/download/9/c/5/9c5b2167-8017-4bae-9fde-d599bac8184a/OS_Desc_Ext_Prop.zip

closed 2019-05-10 by justinmcbride

duncanmacmichael, 2019-05-13:
Hi @justinmcbride, sorry for the delay on this. That should work for the v1.0 spec. Here's a link to the [v2.0 spec](http://download.microsoft.com/download/3/5/6/3563ED4A-F318-4B66-A181-AB1D8F6FD42D/MS_OS_2_0_desc.docx) as well, if you need it. I have an internal work item to get these download pages back up, but it's in my backlog at the moment. I hope these links help you in the meantime.