MicrosoftDocs/windows-driver-docs #1451: Need to Update by Type-C SuperMUTT for WHLK 1809, right?

Deon-Hung, 2019-04-09:
hello, 
USB Type-C ConnEx can be test in WHLK 1803, but fail in WHLK 1809.
The fail content show "no MUTT device connected". 
I found  Type-C SuperMUTT ( https://mcci.com/usb/dev-tools/3501-type-c-supermutt/ ), but the page didn't update yet. If WHLK 1809 need "Type-C SuperMUTT", please update the page!!
Thank a lot!!!!

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 9dbe4bf8-f147-0fbe-2003-cde8fe63df09
* Version Independent ID: 90b89903-686e-940b-4cf5-b7144f4e91d4
* Content: [Microsoft USB Test Tool (MUTT) devices - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/usbcon/microsoft-usb-test-tool--mutt--devices#feedback)
* Content Source: [windows-driver-docs-pr/usbcon/microsoft-usb-test-tool--mutt--devices.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/usbcon/microsoft-usb-test-tool--mutt--devices.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @duncanmacmichael
* Microsoft Alias: **dumacmic**

duncanmacmichael, 2019-04-09:
Hi @Deon-Hung, thanks for opening this issue. I don't have specific details to share right now about a Type-C SuperMUTT doc, but I'll investigate this for you. Stay tuned.

duncanmacmichael, 2019-05-30:
Hi @Deon-Hung, sorry it's been a while since I've checked in. The USB teams are currently heads down developing for the next release of Windows, so I don't have a concrete answer as to when these docs might get published. So, I'm going to go ahead and close this issue.

In the meantime, we do have this Channel 9 video: [Power Delivery Testing with the USB Type C SuperMUTT](https://channel9.msdn.com/Events/WinHEC/WinHEC-Online/Power-Delivery-Testing-with-the-USB-Type-C-SuperMUTT). In the description of that video are additional resources for testing. I also recommend checking the [USB Blog](https://techcommunity.microsoft.com/t5/Microsoft-USB-Blog/bg-p/MicrosoftUSBBlog), including posts like this one: [Running USB Type-C System HLK tests with the Type-C MUTT](https://techcommunity.microsoft.com/t5/Microsoft-USB-Blog/Running-USB-Type-C-System-HLK-tests-with-the-Type-C-MUTT/ba-p/270912).

Hope this helps!

closed 2019-05-30 by duncanmacmichael