MicrosoftDocs/windows-driver-docs #1453: Broken links

kenr, 2019-04-10:
There are lot of broken links on this page. It would be good to update the links.

---
#### Document details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 5cbad219-645b-4478-e284-8136cb65f210
* Version Independent ID: b8628dca-0242-92c3-5830-b0c965c89438
* Content: [Universal Serial Bus (USB) - Windows drivers](https://docs.microsoft.com/en-gb/windows-hardware/drivers/usbcon/)
* Content Source: [windows-driver-docs-pr/usbcon/index.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/usbcon/index.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @duncanmacmichael
* Microsoft Alias: **dumacmic**

duncanmacmichael, 2019-04-10:
Hi @kenr, thanks for pointing this out. It looks like some of those links might have pointed to older pages that got swept up in a content migration we did, so I'll take a look at fixing them.

duncanmacmichael, 2019-04-13:
OK @kenr, I've fixed these links and they should be live in a few minutes. Thanks again for finding these!

closed 2019-04-13 by duncanmacmichael