MicrosoftDocs/windows-driver-docs #1454: Please add a how-to-download/where-to-find section

jrb9249, 2019-04-10:
It would be very useful for me if I could quickly see where to find the pvk2pfx.exe tool; e.g., a download link and/or install instructions.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 7aed1ae0-9fc2-38fe-5879-9db361f574dd
* Version Independent ID: a23498bc-c1fd-4ed7-171c-09d42216fc10
* Content: [Pvk2Pfx - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/pvk2pfx#feedback)
* Content Source: [windows-driver-docs-pr/devtest/pvk2pfx.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/devtest/pvk2pfx.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

EliotSeattle, 2019-04-10:
Hi @jrb9249.
The topic has this info: "A 32-bit version of the Pvk2Pfx tool is located in the bin\x86 folder of the WDK. A 64-bit version of the tool is located in the bin\x64 of the WDK. For example, on an x64-based computer running Windows 10, the path is C:\Program Files (x86)\Windows Kits\10\bin\x64."
On my computer, with multiple versions of the WDK, I see the tool in:
        C:\Program Files (x86)\Windows Kits\10\bin\10.0.17763.0\x64
Hope this helps,
Eliot

closed 2019-04-10 by EliotSeattle

cristiangiagante, 2019-05-20:
Please add the version where this tool can be downloaded it not exist in all win sdk versions.
Please upload it !

csrowell, 2021-01-05:
I know this is a different SDK, but for those who land here from Google, I was able to find the file at C:\Program Files (x86)\Windows Kits\10\bin\10.0.19041.0\x86 after installing the latest Windows Software Development Kit for Windows 10, making sure to choose "Windows SDK for Desktop C++ x86 apps". The grandparent directory's name will differ depending on what version of the SDK you get.