MicrosoftDocs/windows-driver-docs #1456: Incorrect documentation

bslava, 2019-04-11:
From my experience "devcon dp_delete -f inf" won't work. The correct command is:
devcon -f dp_delete inf

-f parameter must be placed before dp_delete

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: e9f37b92-708c-e826-fcf4-e663d35759d3
* Version Independent ID: 0cf2a07f-f355-08bf-d912-d7ace4be3d75
* Content: [DevCon Dp_delete - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/devcon-dp-delete#feedback)
* Content Source: [windows-driver-docs-pr/devtest/devcon-dp-delete.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/devtest/devcon-dp-delete.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

DOMARS, 2019-04-11:
@EliotSeattle - This PR looks good, the flag comes first as shown in the syntax block and online help.
```
C:\Program Files (x86)\Windows Kits\10\Tools\x64>devcon help dp_delete
devcon [-f] dp_delete <inf>
Deletes a third-party (OEM) driver package.
This command will only work on the local machine.
[-f] will force delete the driver package, even
if it is in use by a device.
<inf> is the name of a published INF on the local
machine.  This is the value returned from dp_add
and dp_enum.```

EliotSeattle, 2019-04-11:
Thanks, @DOMARS, for the review. And thank you, @bslava, for the PR!!

EliotSeattle, 2019-04-11:
Fixed in source. Will be live shortly

closed 2019-04-11 by EliotSeattle

bslava, 2019-04-14:
Thanks for the fix! I think it is necessary to fix the "Sample Usage" section as well.