MicrosoftDocs/windows-driver-docs #1457: Instructions needed for multi-arch driver submissions

jkunkee, 2019-04-11:
If each folder needs to support the same set of architectures, how does one folder support multiple architectures? It seems from the examples here that each folder can only have one INF and one SYS in it.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: ed287852-04b7-1569-fd51-2640cfa24f57
* Version Independent ID: a8ce732f-5b41-f1ce-e37d-a7c2794546a4
* Content: [Attestation signing a kernel driver for public release - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/attestation-signing-a-kernel-driver-for-public-release#feedback)
* Content Source: [windows-driver-docs-pr/dashboard/attestation-signing-a-kernel-driver-for-public-release.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/dashboard/attestation-signing-a-kernel-driver-for-public-release.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

RAJU2529, 2019-11-09:
@jkunkee. Before installing the driver software, I usually extract the contents, I have seen both the folders namely Win32 and AMD64 or x64. But only only .inf file contains both x86 x64 scripts.  But two separate .sys files, for example,
avidmedia32.sys
avidmedia64.sys

tedhudek, 2020-11-16:
Refers to Note in https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/attestation-signing-a-kernel-driver-for-public-release#create-a-cab-files-submission

tedhudek, 2020-11-16:
Thanks for the question and apologies for the delay.  I think the solution here is to specify multiple architectures in the INF.  Please let me know if this remains an issue and that doesn't help!

closed 2020-11-16 by tedhudek