MicrosoftDocs/windows-driver-docs #1462: this registry works for windows10 RS3?

garnid, 2019-04-13:
Will this registry key works on latest revisions like windows 10 RS3? Whats the default VSync idle time-out value in widows 10 RS3?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 9d60189d-749b-77dc-cbde-39e1caa28fb5
* Version Independent ID: 2729358b-09f1-14f6-b9b0-362073433afd
* Content: [Saving Energy with VSync Control - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/display/saving-energy-with-vsync-control#feedback)
* Content Source: [windows-driver-docs-pr/display/saving-energy-with-vsync-control.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/display/saving-energy-with-vsync-control.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @mimisasouvanh
* Microsoft Alias: **misaso**

garnid, 2019-04-25:
where can I check for the answer?  answer will be posted here itself?

lorihollasch, 2019-10-15:
@garnid, sorry for the slow response. I've updated the page to clarify OS support. The registry control information applies to Windows Vista SP1 and later versions of the Windows OS (which includes Windows 10 RS3).

closed 2019-10-15 by lorihollasch

IshaniKam, 2020-03-13:
Sorry for reviving a closed thread.            
The regkey does not seem to work on RS5. 
Has there been any recent change in the regkey name or path?