MicrosoftDocs/windows-driver-docs #1463: devcon dp_enum command does not supply enough data to detect device

bslava, 2019-04-14:
I need your help. For my device "devcon dp_enum" command prints the following:

oem4.inf
    Provider: Intel
    Class: Universal Serial Bus controllers

The supplied data does not allow detection the device unambiguously. There can be several different devices that share the same provider and class. What should I do to get the correct oem*.inf name?

By the way, "devcon dp_enum" does not print "Data" and "Version" fields as shown on the example above.

Thanks in advance.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 0059a230-b9ed-8dae-c111-9849586d88cc
* Version Independent ID: 289c1527-7af4-2c26-cf44-910ba72e589c
* Content: [Example 45 Add and Remove Driver Packages - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/example-45--add-and-remove-driver-packages)
* Content Source: [windows-driver-docs-pr/devtest/example-45--add-and-remove-driver-packages.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/devtest/example-45--add-and-remove-driver-packages.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

shivaprasadh, 2019-05-17:
Having the same question.  Can someone help to answer this question?

DOMARS, 2021-01-26:
@bslava and @shivaprasadh - Thanks for the interest in the Windows Docs.  This particular channel is monitored by the men and women who write the driver documentation posted on docs.microsoft.com/drivers. Though we'd like to, we don’t have the resources to help with the unique challenges of hunting down a specific OEM Inf file.

A few tips that may help.

You can look in device manager and locate the Pnp ID. This can be helpful in tracking down more information. For example I can use the Pnp ID and the devcon driverfiles command, like this.

```
C:\Program Files (x86)\Windows Kits\10\Tools\x64>devcon driverfiles "usb\vid_045E&PID_0812*"
USB\VID_045E&PID_0812&MI_00\8&186D3932&1&0000
    Name: Microsoft« LifeCam Cinema(TM)
    Driver installed from C:\WINDOWS\INF\usbvideo.inf [USBVideo]. 2 file(s) used by driver:
        C:\WINDOWS\system32\DRIVERS\usbvideo.sys
```

Perhaps more directly, you can use the file explorer or various utilities to search the text of the OEM files in the INF directory for a string that may identify the OEM inf file. Then using the inf file shown in the driverfiles output, I can compare that to the candidate OEM Inf files identified in searching the text of the OEM INF files.

If it's still an issue, you may have success posting your question in the Q&A forums for Windows, where other driver developers can be found. 
https://docs.microsoft.com/en-us/answers/products/windows
 
Again, thanks for reaching out. Best of luck with tracking down the OEM inf file.

closed 2021-01-26 by DOMARS