MicrosoftDocs/windows-driver-docs #1472: Where is it shown in a running server which build type is running?

arfyness, 2019-04-17:
This article omits any information about determining what build type is running.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: e1712291-4cb2-7a3f-1298-9c68cb05625d
* Version Independent ID: 41defd10-d145-c65c-1145-b936bc10913a
* Content: [Checked and Free Build Differences - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/checked-and-free-build-differences#feedback)
* Content Source: [windows-driver-docs-pr/devtest/checked-and-free-build-differences.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/devtest/checked-and-free-build-differences.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

kml27, 2019-09-26:
https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/installing-just-the-checked-operating-system-and-hal--for-windows-vist#step-4---restart-the-computer

EliotSeattle, 2019-09-27:
Thanks for the Issue, @arfyness 

Added the final paragraph in https://docs.microsoft.com/windows-hardware/drivers/devtest/checked-and-free-build-differences#feedback that directs to the info about discovering which version is used.

closed 2019-09-27 by EliotSeattle