MicrosoftDocs/windows-driver-docs #1474: Can't download insider WDK

wjk, 2019-04-18:
Whenever I click the link to download the insider WDK, I get a spinning wait screen for ~10 seconds, then the following message:

&gt; We encountered a problem with your request. Please visit Microsoft Support Contact Us page for assistance. Refer to Message Code 715-123130 and Transaction ID c0e72008-944a-4283-93e0-c88df8bf5cd4.

This has been occurring for at least several months now. This occurs with both the "regular" (installed) and Enterprise WDKs. I would appreciate it if whatever problem is causing this error could be fixed. Thanks!

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 34d06a09-bca8-3177-00fd-2195f56c4fb9
* Version Independent ID: 5328b553-ad08-4808-0b14-9720484040ce
* Content: [Installing preview versions of the Windows Driver Kit (WDK) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/installing-preview-versions-wdk#feedback)
* Content Source: [windows-driver-docs-pr/installing-preview-versions-wdk.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/installing-preview-versions-wdk.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

EliotSeattle, 2019-04-18:
Thanks, @wjk, for filing the issue. We are looking into this. So far, though, we are unable to reproduce this. Would you mind telling me about the computer you see this issue on? OS, processor, do you have other WDK builds already installed, etc?
Thanks,
Eliot

wjk, 2019-04-18:
My OS is Windows 10, build 17763.437. My processor (per msinfo32) is "Intel(R) Core(TM) i5-6400T CPU @ 2.20GHz, 2208 Mhz, 4 Core(s), 4 Logical Processor(s)". I am using Edge Insider Preview (dev) version 74.1.96.24, although this issue happened with the UWP-based "old" Edge as well. The only WDK build I have installed is Enterprise WDK 1809, although I don't see how that would affect the browser-based download process. Let me know if there's anything else you need. Thanks!

EliotSeattle, 2019-04-18:
Got it, @wjk.
It looks like there was some behavior that triggered a block on downloads. I believe that the team has identified the cause and reset the mechanism. Can you try to download the kit again? You should be good to go.
Eliot

wjk, 2019-04-18:
Downloading now. Thanks so much!

closed 2019-04-18 by wjk

EliotSeattle, 2019-04-18:
Excellent!!! Thanks for bringing this to our attention.