MicrosoftDocs/windows-driver-docs #1476: What does "Netmon will not be carried" implies ?

gpotter2, 2019-04-18:
Hello,

https://github.com/MicrosoftDocs/windows-driver-docs/blob/05ab17d20257649dd445ce964e2128de24a89c05/windows-driver-docs-pr/network/wdi-features-not-carried-over-in-wdi.md says that `NetMon` will not be carried over to WDI.

Does this mean that it won't be possible to let say, perform a Raw WiFi capture anymore once the deprecated API is cropped (which I assume will happen someday) ?

I actually bumped into this because of `https://github.com/nmap/nmap/issues/1528` that pointed towards here. This feature was used by `Npcap` (successor of `Winpcap` on which many projects rely on today), and allowed to receive (/send) custom 802.11 frames. It allowed to make a lot of tools based on Network monitoring.

Would you have any extra informations about what the alternatives are, or what exactly is deprecated & will be removed ?

Thanks for the support !

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: aa1ea60a-9358-5b36-65c7-f3299c6af50f
* Version Independent ID: 2b8d3a5f-ae58-8ef9-a83d-2dc4d8f87393
* Content: [Features not carried over in WDI - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/network/wdi-features-not-carried-over-in-wdi#feedback)
* Content Source: [windows-driver-docs-pr/network/wdi-features-not-carried-over-in-wdi.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/network/wdi-features-not-carried-over-in-wdi.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @duncanmacmichael
* Microsoft Alias: **dumacmic**

duncanmacmichael, 2019-04-18:
Hi @gpotter2, great question! I'm not sure as to the full answer on this, but I'll check with the Wi-Fi team and see what they have to say. Stay tuned. :)

gpotter2, 2019-04-22:
Thanks for the support, I'll be looking forward to it :-)

guyharris, 2019-04-29:
> I actually bumped into this because of some concerns shared by a maintainer of libpcap on [nmap/nmap#1528](https://github.com/nmap/nmap/issues/1528) that pointed towards here. This feature was used by `Npcap` (successor of `Winpcap` on which many projects rely on today), and allowed to receive (/send) custom 802.11 frames. It allowed to build a lot of tools based on Network monitoring.

I'm the maintainer in question, and I added support to libpcap for turning monitor mode on and off; it's supported on several UN\*Xes (although some UN\*Xes have the annoying habit of changing the way it's done, so the Linux and FreeBSD support need to be updated).

It's used by, for example, tcpdump and Wireshark for doing passive sniffing on 802.11 networks.

gpotter2, 2019-06-02:
Hi @duncanmacmichael & sorry for the ping.

Did you have a chance to ask the Wi-Fi team about this ?

Thanks !!

duncanmacmichael, 2019-06-03:
No worries @gpotter2; I do apologize that this slipped under my radar recently. I did get an answer from a Wi-Fi team member but mentally filed it away thinking I was waiting for further info when in fact the answer was there. Sorry about that!

Netmon was not carried over to WDI as it did not fall into the category of required functionality when the Wi-Fi driver team was making design decisions for WDI. In addition, the team discovered that many drivers did not implement this well enough to make it reliable. In a related vein, the Native 802.11 miniport driver model was indeed officially deprecated with the release of Windows 10, which supports WDI instead.

As for alternatives, I don't know of any personally but the team did say this:

"Currently, we expect customers to use the available commercial solutions, but if there is sufficient ask for it, we can consider it for the next driver model update."

So, I'd recommend submitting a feature request through the Feedback Hub or through the "This Product" feedback button on this page. However, keep in mind that they did say that they would consider this for the next _driver model_ update, not the next WDI update, and the timeline of that is completely unknown at this point.

Hope this helps!

closed 2019-06-03 by duncanmacmichael

gpotter2, 2019-06-10:
Thanks for the answer !

yujuiwu, 2021-08-16:
Hi @duncanmacmichael,

Not sure if it's feasible to ask here, but  any idea whether Netmon is going to be supported in Windows 11? 

It would be great if we can capture WiFi traffic over Wireshark on Windows.

Thanks!

guyharris, 2021-08-16:
> Not sure if it's feasible to ask here, but any idea whether Netmon is going to be supported in Windows 11?

Note that full "support" might involve parties other than Microsoft.  As @duncanmacmichael said:

> In addition, the team discovered that many drivers did not implement this well enough to make it reliable.

Microsoft are not, as far as I know, responsible for all Wi-Fi adapter drivers on Windows, so to ensure adequate "support" would require one of:

- Microsoft taking over full responsibility for those drivers, and making sure they all support monitor mode well;
- Microsoft requiring good monitor mode support in drivers for hardware that can support monitor mode, and making that part of the WHQL testing, so if your driver has a problem with monitor mode, you don't get qualified;
- somebody maintaining a list of drivers and the quality of their monitor mode support, so that people at least know what adapters don't provide adequate support.

Adding support for monitor mode to WDI without some form of quality assurance for monitor mode support might just move Windows from "many drivers either don't support monitor mode under Native Wi-Fi or do a bad job of it" to "many drivers either don't support monitor mode under WDI or do a bad job of it".