MicrosoftDocs/windows-driver-docs #1477: win10s driver targeting on Windows update

LiaoMichelle, 2019-04-19:


LiaoMichelle, 2019-04-19:
Base and Extension driver inf are target on Windows update for Windows 10 in S mode.
But only extension driver inf can be downloaded on non-Win10 S OS.
Is that behavior correct?

tedhudek, 2019-05-17:
Hi @LiaoMichelle thanks for the feedback and apologies for the delay.

Can you please clarify your question?  You have a base INF and extension INF, both targeting Windows 10 in S mode.  Are you asking if the extension INF will be installed on 10 S machines?

LiaoMichelle, 2019-05-20:
Hi Ted,
I am asking why extension INF was installed  on "Non" win10 S machines?
Thanks

tedhudek, 2019-05-20:
So you are targeting Windows 10 S, and base and extension INF are installed on 10 S successfully, and in addition, extension INF (but not base INF) is installed on Win10 desktop?

LiaoMichelle, 2019-05-21:
Yes, in addition, extension INF (but not base INF) is installed on Win10 desktop.

tedhudek, 2019-05-24:
What is your desired result?  Do you want both base and extension installed on Win10 desktop?  Or neither?

tedhudek, 2019-05-24:
@LiaoMichelle I've added two sections to https://docs.microsoft.com/en-us/windows-hardware/drivers/install/windows10sdriverrequirements that should help with your problem.  Please see **Installation** and **Troubleshooting installation** on that page.  Closing the issue for now, please reopen if needed.

closed 2019-05-24 by tedhudek