MicrosoftDocs/windows-driver-docs #1482: Windbg .foreach command should specify that initialSkipNumber is in Hex

adrianvmsft, 2019-04-23:
Windbg .foreach command should specify that initialSkipNumber is in Hex

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 84d76278-1ba1-07e8-bada-6f66484f9330
* Version Independent ID: db7baa82-2eed-80a6-8b01-8134d5553744
* Content: [.foreach - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/-foreach#feedback)
* Content Source: [windows-driver-docs-pr/debugger/-foreach.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/-foreach.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2019-05-02:
@adrianvmsft - Could you provide an example usage of .foreach where hex is used? I'm wondering if for certain commands, another numbering system would be used.

adrianvmsft, 2019-05-02:
I used it in combination with DumpArray. Here is an example:
https://chris.eldredge.io/blog/2014/03/05/windbg-dump-entries-in-array/

That /pS 16 is in Hex. So it skips 22 (decimal) initial tokens.

I tried it using
.foreach /pS 16 (addr {!DumpArray /d 301d6910}) {.printf "${addr}\n"}

I specified various values starting with 0, 1, ... 9 to confirm the starting token. When you jump from 9 to 10 it skips much more because it is in hex. So you need to use A, B, C, D, E, F, 10, etc..

DOMARS, 2019-05-09:
@adrianvmsft - This may be related to your base number system (radix). You can use the n command to display and change the current system wide radix.  

```
0:000> n
base is 16
```

Also relevant here are the topics related to numbering such as [MASM Numbers and Operators] (https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/masm-numbers-and-operators) describes how to prefix with 0x for hex and 0n to force radix. I have not tested that with .foreach
Also [https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/c---numbers-and-operators](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/c---numbers-and-operators)

I personally would consider WinDbg preview and the JavaScript environment these days for this type of work. See - [WinDbg Preview - Scripting](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/windbg-scripting-preview)

DOMARS, 2019-06-05:
@adrianvmsft - I believe this behavior is related to the base number system (radix) as described in the comment above. For that reason, I don't plan on updating the topic to describe this behavior right now. Thank you for passing on what you are seeing with the .foreach command and hex numbering.

closed 2019-06-05 by DOMARS