MicrosoftDocs/windows-driver-docs #1485: ResourceToRelease is not to be released here

alfaunits, 2019-04-24:
IIRC, this is a pointer to a variable that will receive a pointer to the ERESOURCE structure, that should be passed in RELEASE_FOR_MOD_WRITE (not by the minifilter itself, but the caller).
The resource is not released in this call!

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: dba5be64-293b-e562-0839-7dc4d3790908
* Version Independent ID: 29da14fe-0b56-a36a-bb19-db79d5ce801c
* Content: [FLT_PARAMETERS for IRP_MJ_ACQUIRE_FOR_MOD_WRITE union - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ifs/flt-parameters-for-irp-mj-acquire-for-mod-write)
* Content Source: [windows-driver-docs-pr/ifs/flt-parameters-for-irp-mj-acquire-for-mod-write.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/ifs/flt-parameters-for-irp-mj-acquire-for-mod-write.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

rodwiddowson, 2019-04-24:
FWIW:

I have the scars that it absolutely *MUST* be set to something if you return FLT_COMPLETE.  Trust me, you'll not regret it

I have a vague memory that in the world without filter managers the resource *IS* release iff you do not specify the MJ_PRE_RELEASE_FOR_MODWRITER.  Something (FAT, or maybe PINBALL) did exactly that...

lorihollasch, 2019-07-18:
@alfaunits , really great catch! I've updated the page to note that ResourceToRelease is actually being acquired here (and is indeed a pointer to a pointer), and will be released in IRP_MJ_RELEASE_FOR_MOD_WRITE. Thanks for reporting these errors.