MicrosoftDocs/windows-driver-docs #1486: Availabilty of checked builds

alfaunits, 2019-04-24:
Not an issue with the documentation, but since I have found no kernel dev who was able to install Windows 10 CHecked build from the MSDN, it is prudent to provide help during installation.

Or fix the installation :D

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: c654fb8a-3fed-aa30-7535-62059b6a6a39
* Version Independent ID: 11fc0183-2772-bc79-84a5-8d52d084aae9
* Content: [Checked Build of Windows - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/checked-build-of-windows)
* Content Source: [windows-driver-docs-pr/devtest/checked-build-of-windows.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/devtest/checked-build-of-windows.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

alfaunits, 2019-11-01:
Nothing on this still?

EliotSeattle, 2019-11-04:
Still hunting for an answer. Apologies for the delays.

DOMARS, 2020-05-09:
@alfaunits - As it turns out, checked builds of Windows are no longer distributed. Checked builds are only available on older versions of Windows, before Windows 10 version 1803. So I added a note about this. Instead of checked builds, use tools such as Driver Verifier and GFlags to check driver code in later versions of Windows. You can continue to use debug builds of your own driver code. Thanks for filing this issue and sharing what you were seeing, and the interest in improving our docs.

closed 2020-05-09 by DOMARS

alfaunits, 2020-05-09:
Actually, NONE of the Windows 10 has checked builds, except 1511.
And noone succeeded in installing that build.

DOMARS, 2020-05-09:
@alfaunits - OK, thanks for sharing your experiences with checked builds and Windows 10. As a number of years have passed and in the past we had multiple distribution methods for Windows beyond the MSDN download, it would be a bit difficult to provide more precision. But I hope that this note will communicate that they are no longer available with current day Windows.

biorpg, 2021-04-06:
> 
> Actually, NONE of the Windows 10 has checked builds, except 1511. And noone succeeded in installing that build.

I was able to install it by altering the system time so that the certificates weren't rejected by the installer.