MicrosoftDocs/windows-driver-docs #1490: BugCheck(139) parameter 1= 0x1e , what does it mean?

andychenami, 2019-04-25:
[Enter feedback here]
I got a BugCheck(139) random happend during platform cold boot or warm boot
The parameter 1 is 1e, but I can not get his description from list.
May I ask what does it mean?
The attached is the minidump message.

Many thanks.
Andy Chen

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: f1073c44-369a-9365-a2a7-ea5b064fb1af
* Version Independent ID: 28187f46-abb9-c7cb-9b22-ddd4a2ec9d37
* Content: [Bug Check 0x139 KERNEL_SECURITY_CHECK_FAILURE - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x139--kernel-security-check-failure)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0x139--kernel-security-check-failure.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0x139--kernel-security-check-failure.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2019-10-09:
@andychenami - I checked with the dev team and these additional codes are for Windows internal use, so I will not be able to document them. Thanks for passing on the parameter 1 value that you encountered and thanks for your interest in improving the stop code documentation.

closed 2019-10-09 by DOMARS

HollisTech, 2020-05-06:
Please document all of the bugcheck parameters values. Outside microsoft it is really difficult to analyze crashes when the information that is made available is undocumented. It is hard enough when the documentation is available. "These additional codes are for Windows internal use" is not 'closing the issue', it is avoiding the issue.