MicrosoftDocs/windows-driver-docs #1495: Add documentation regarding PCI Express root port and endpoint device power state behavior

tzimmer1, 2019-04-26:
Looking to see if there is any documentation in flight about PCI Express root port and endpoint device power state (Dx) behavior.
My understanding is that Microsoft has given a commitment to Intel that this will be done in the very near future.
aadake@microsoft.com may have more current information.  Thanks.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 874abd8a-91da-1c1b-4446-711e18f3092c
* Version Independent ID: f66dd88a-d1b0-87e3-2c9c-e9aa723d591f
* Content: [PCI Power Management and Device Drivers - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/pci/pci-power-management-and-device-drivers#feedback)
* Content Source: [windows-driver-docs-pr/pci/pci-power-management-and-device-drivers.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/pci/pci-power-management-and-device-drivers.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

tzimmer1, 2019-06-05:
Closing.   We have filed this as a workitem between Intel and Microsoft in a Partner Area.
Supposedly it is coming soon from aadake@microsoft.com.

We will see...

tzimmer1, 2019-07-16:
Reopening as Intel has received to traction on our work item.
Maybe this item will give traction with Aacer.

EliotSeattle, 2019-07-16:
Hi @tzimmer1 
We do not normally discuss future plans for features or content in our documentation. I do not have documentation about PCI Express root port and endpoint device power state (Dx) behavior, but if that feature does get released, we will document it as close as we can to the time it is included in an insider preview release.

Sorry that I cannot offer more at this time.

Eliot

closed 2019-07-16 by EliotSeattle