MicrosoftDocs/windows-driver-docs #1497: This is already diffrent fromt the portal

puffelyang, 2019-04-29:
This document is already misalign with the portal. The procedure, screenshot... are not correct anymore.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: ead14b61-61e5-d45f-d5c9-d3d4c0bcf93d
* Version Independent ID: 31c51703-a968-0a6a-bba0-bdb47619b018
* Content: [Hardware dashboard API - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/dashboard-api#feedback)
* Content Source: [windows-driver-docs-pr/dashboard/dashboard-api.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/dashboard/dashboard-api.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

EliotSeattle, 2019-04-29:
Thanks, @puffelyang. 
I'll look into the discrepancies and update. If you could let me know of any in particular, that'd be very helpful.

puffelyang, 2019-04-30:
Thank you for your reply, Eliot.

I’ve experienced trouble to locate the user setting since the setting “menu” is clearly changed, and you’ll need to goes into the “Developer settings” to do the rest of the settings for user. However, this is just my experience, and I didn’t follow the document step by step, so there could also be something get skipped or ignored. In that sense, I’d suggest that you do follow your document step by step and you’ll found out the change on the portal, and maybe other change that I’ve overlooked.

In addition, this is the part I don’t like the “DevOps” ideology, that the changes indeed happened more quickly and easily, but also likely to be published with less considering of other linked items, since anything can be “easily” changed or updated, such a document in this case. Things are getting smaller, easier and less lost, together with more discrete and less responsibility to each other. Thinking about the Boeing 737MAX.

Hum… ok, maybe this is too much.

Good luck, and really hope that documents can always been properly updated.

Regards!
Weishan

EliotSeattle, 2019-04-30:
Thanks for taking the time to let me know about this, Weishan. It seems like some things have changed in the Partner Center dashboard UI for submissions that I wasn't aware of. I will get together as soon as possible with the PM for the Partner Center to walk through and update our content. We normally do walk through each page to make sure the docs and the portal are the same. Obviously, we got out of sync in the last few weeks. Hopefully, this won't have nearly the same catastrophic outcome as an airline engineering mistake, tough!

Thanks for bringing this to my attention. I apologize for any hardship and appreciate your help. I will keep this and Issue #1502 open to report progress.

tedhudek, 2020-11-10:
Need to review UI steps in https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/dashboard-api#assigning-the-appropriate-hardware-roles-to-your-azure-ad-application to see if they match current reality

tedhudek, 2020-11-16:
Thanks for the query, @puffelyang, and apologies for the delay.

If this is still an issue, I would recommend looking in the Azure documentation.  Here's a possible starting point -- https://docs.microsoft.com/azure/active-directory/roles/delegate-by-task.

closed 2020-11-16 by tedhudek