MicrosoftDocs/windows-driver-docs #1499: Missing Parameter 0x6001

Sparkx120, 2019-04-29:
I hit a 0x6001 while doing my usual work in Windows 10. Here is the Event Log

- &lt;Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event"&gt;
- &lt;System&gt;
  &lt;Provider Name="Microsoft-Windows-Kernel-Power" Guid="{331c3b3a-2005-44c2-ac5e-77220c37d6b4}" /&gt; 
  &lt;EventID&gt;41&lt;/EventID&gt; 
  &lt;Version&gt;6&lt;/Version&gt; 
  &lt;Level&gt;1&lt;/Level&gt; 
  &lt;Task&gt;63&lt;/Task&gt; 
  &lt;Opcode&gt;0&lt;/Opcode&gt; 
  &lt;Keywords&gt;0x8000400000000002&lt;/Keywords&gt; 
  &lt;TimeCreated SystemTime="2019-04-25T04:34:10.676151000Z" /&gt; 
  &lt;EventRecordID&gt;37649&lt;/EventRecordID&gt; 
  &lt;Correlation /&gt; 
  &lt;Execution ProcessID="4" ThreadID="8" /&gt; 
  &lt;Channel&gt;System&lt;/Channel&gt; 
  &lt;Computer&gt;ZenMachine&lt;/Computer&gt; 
  &lt;Security UserID="S-1-5-18" /&gt; 
  &lt;/System&gt;
- &lt;EventData&gt;
  &lt;Data Name="BugcheckCode"&gt;26&lt;/Data&gt; 
  &lt;Data Name="BugcheckParameter1"&gt;0x6001&lt;/Data&gt; 
  &lt;Data Name="BugcheckParameter2"&gt;0x0&lt;/Data&gt; 
  &lt;Data Name="BugcheckParameter3"&gt;0x0&lt;/Data&gt; 
  &lt;Data Name="BugcheckParameter4"&gt;0x0&lt;/Data&gt; 
  &lt;Data Name="SleepInProgress"&gt;0&lt;/Data&gt; 
  &lt;Data Name="PowerButtonTimestamp"&gt;0&lt;/Data&gt; 
  &lt;Data Name="BootAppStatus"&gt;0&lt;/Data&gt; 
  &lt;Data Name="Checkpoint"&gt;41&lt;/Data&gt; 
  &lt;Data Name="ConnectedStandbyInProgress"&gt;false&lt;/Data&gt; 
  &lt;Data Name="SystemSleepTransitionsToOn"&gt;9&lt;/Data&gt; 
  &lt;Data Name="CsEntryScenarioInstanceId"&gt;0&lt;/Data&gt; 
  &lt;Data Name="BugcheckInfoFromEFI"&gt;true&lt;/Data&gt; 
  &lt;Data Name="CheckpointStatus"&gt;0&lt;/Data&gt; 
  &lt;/EventData&gt;
  &lt;/Event&gt;

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: de955911-dfbf-33b3-778a-767a5a578a0d
* Version Independent ID: 23fb5cbd-5f7d-f7ca-1622-5885020966c2
* Content: [Bug Check 0x1A MEMORY_MANAGEMENT - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x1a--memory-management)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0x1a--memory-management.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0x1a--memory-management.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2019-06-21:
@Sparkx120  - I have draft text prepared and it is being reviewed by the dev team for this parameter 1 value for bug check 0x1A.

DOMARS, 2019-07-01:
@Sparkx120 - The description for this bug check 0x1A parameter 1 is live. Thanks for your interest in improving the stop code documentation.

closed 2019-07-01 by DOMARS