MicrosoftDocs/windows-driver-docs #1500: Additional stop code number

LuigiBruno, 2019-04-29:
This kind of bugcheck is often recorded in the Windows Event Viewer with the code 278, as per example below
&lt;EventData&gt;
    &lt;Data Name="BugcheckCode"&gt;278&lt;/Data&gt;
    &lt;Data Name="BugcheckParameter1"&gt;0xffffc4828b4d6010&lt;/Data&gt;
    &lt;Data Name="BugcheckParameter2"&gt;0x0&lt;/Data&gt;
    &lt;Data Name="BugcheckParameter3"&gt;0x0&lt;/Data&gt;
    &lt;Data Name="BugcheckParameter4"&gt;0x0&lt;/Data&gt;
…
&lt;/EventData&gt;

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 367e4ff4-f83e-b642-095b-27769fb05505
* Version Independent ID: 24848013-3432-4ae4-8ffd-b664aefdd069
* Content: [Bug Check 0x116 VIDEO_TDR_FAILURE - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x116---video-tdr-failure)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0x116---video-tdr-failure.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0x116---video-tdr-failure.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2019-06-21:
@LuigiBruno - This makes sense as 258 is decimal representation of 0x116. If you like, submit a Windows Product Feedback item to have this value display in hex in the event log.

closed 2019-06-21 by DOMARS

LuigiBruno, 2019-06-25:
> @LuigiBruno - This makes sense as 258 is decimal representation of 0x116. If you like, submit a Windows Product Feedback item to have this value display in hex in the event log.

Thanks @DOMARS.