MicrosoftDocs/windows-driver-docs #1501: More parameter 1 codes?

englishb, 2019-04-29:
The article says that parameter 1 identifies the exact violation, but only codes 0x1 - 0x4 are listed.  I have run into a bugcheck 0x119 with parameter 1 set to 0x400.  What does this code mean?  I believe it could be related to our video driver, but the lack of documentation on this value doesn't really give me any leads.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 7917857b-97bc-efec-67ab-5e3af061524c
* Version Independent ID: 8c00d20e-9b21-7bb0-a041-220aca9d2487
* Content: [Bug Check 0x119 VIDEO_SCHEDULER_INTERNAL_ERROR - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x119---video-scheduler-internal-error)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0x119---video-scheduler-internal-error.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0x119---video-scheduler-internal-error.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

helgeklein, 2019-04-30:
I just got this with parameter 1 set to the value 5.

DOMARS, 2020-02-11:
@englishb - I added a description for the 0x400 parameter. Thank for this issue and your interest in the debugger stop code documentation for driver developers.

closed 2020-02-11 by DOMARS