MicrosoftDocs/windows-driver-docs #1502: This document is already old and mis-align with the new Azure portal now.

puffelyang, 2019-04-30:
As subject. 

Now, in the "App registration (Legacy)", and some settings are not able to be found anymore. 

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: dd5fde9f-4ce1-d962-1b65-241f2fbb3fd3
* Version Independent ID: fb559713-fd60-4f32-e09b-f34c77ecc5de
* Content: [Enable secure data sharing - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/enable-secure-data-sharing)
* Content Source: [windows-driver-docs-pr/dashboard/enable-secure-data-sharing.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/dashboard/enable-secure-data-sharing.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **shganesh**

EliotSeattle, 2019-04-30:
Similar to Issue #1497

tedhudek, 2020-11-10:
Page in question now redirects to https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/dashboard-api

closed 2020-11-10 by tedhudek