MicrosoftDocs/windows-driver-docs #1504: [LOC] "アプリ & 機能" needs to be replaced with "アプリと機能"

hyoshioka0128, 2019-04-30:
https://docs.microsoft.com/ja-jp/windows-hardware/drivers/debugger/debugger-download-tools

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 162fe2fe-9216-f4c5-9f91-3df8776bb99e
* Version Independent ID: 013dbf6b-0790-9c11-24f4-dbf24d64401e
* Content: [Download Debugging Tools for Windows - WinDbg - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/debugger-download-tools)
* Content Source: [windows-driver-docs-pr/debugger/debugger-download-tools.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/debugger-download-tools.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2019-05-02:
@EliotSeattle - I think this is a localization issue. Can you provide guidance on how to best direct this request if we are setup to handle this?

EliotSeattle, 2019-05-02:
Reached out to the loc team to see how we handle this process. We have steps in place for the loc team to reach out to the core doc team, but not the other way around. Will add comments as I learn more.

EliotSeattle, 2019-06-06:
@v-nidolc 
Can you please route this to the right person?

v-nidolc, 2019-07-12:
@EliotSeattle can you assign the issue to davidprio? Then our vendor will process it. Thanks!

EliotSeattle, 2019-09-10:
@DavidPrio has this been taken care of?
Thanks,
Eliot

hyoshioka0128, 2019-09-11:
@EliotSeattle - Thanks.
Fixd.
Close #1504

closed 2019-09-11 by hyoshioka0128