MicrosoftDocs/windows-driver-docs #1511: Unable to install WDK with VisualStudio 20019

tyszja, 2019-05-01:
According to the [instructions](https://docs.microsoft.com/en-us/windows-hardware/drivers/download-the-wdk#feedback) it should be enough to install VS2019 with C++ support for Windows desktop apps before installing the WDK, but the installer refuses to continue due to missing compatible SDK and points me to the SDK installer. Visual Studio installer shows that I already have the Windows 10 SDK installed. What am I missing? Do I need to download additioanal ~3gigs of SDK in a different location to get the WDK installer to work or is this a different Windows SDK than the VS2019 has? Do I need any other SDKs to get the driver examples to compile?

Regards.

tedhudek, 2019-05-30:
Feedback pertains to https://docs.microsoft.com/en-us/windows-hardware/drivers/download-the-wdk#feedback.

EliotSeattle, 2019-06-13:
We've updated the download process to avoid troubles like this. Sorry about the delay.

closed 2019-06-13 by EliotSeattle