MicrosoftDocs/windows-driver-docs #1531: Resolution steps needed

LuigiBruno, 2019-05-06:
Add analysis and resolution steps.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: cb94bb71-2a31-b7e6-8313-f387ebfd3c53
* Version Independent ID: 95dafa9f-2dda-dc1d-b905-84a87488f708
* Content: [Bug Check 0x193 VIDEO_DXGKRNL_LIVEDUMP - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x192--video-dxgkrnl-livedump#feedback)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0x192--video-dxgkrnl-livedump.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0x192--video-dxgkrnl-livedump.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2020-02-27:
@LuigiBruno - Thanks for the input that we should add additional information on this Bug Check. We use customer input like this as well as data on how often the bug checks occur to determine how we allocate our documentation team resources.

For additional tips on troubleshooting, if you have active issue, I would recommend the forums, for example the WDK one:

•	[Windows Hardware WDK and Driver Development](https://social.msdn.microsoft.com/Forums/windowsdesktop/en-US/home?forum=wdk)

LuigiBruno, 2020-03-04:
> @LuigiBruno - Thanks for the input that we should add additional information on this Bug Check. We use customer input like this as well as data on how often the bug checks occur to determine how we [...]

Thanks @DOMARS.

DOMARS, 2020-12-18:
@LuigiBruno - I'm opened an internal work item to track the work to provide general information on LIVEDUMPs such as this one for the video systems in Windows, so closing this issue. Thanks for passing on what you are seeing. We use this and other data to help drive the work that we do!

closed 2020-12-18 by DOMARS