MicrosoftDocs/windows-driver-docs #1542: Cannot find Printer Driver V4 in Visual Studio 2019

Voidgear, 2019-05-09:
When I click "Create a new project" and search for "printer", I didn't any result coming up. Do I need to install any extra packages? You earliest respond is much appreciated. 

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 227ceb38-396a-aa19-6304-be3df9b0ca08
* Version Independent ID: 2f34593f-aa20-384a-b1bb-c8a962ec4706
* Content: [Building a Basic v4 Printer Driver - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/print/building-a-basic-v4-printer-driver#)
* Content Source: [windows-driver-docs-pr/print/building-a-basic-v4-printer-driver.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/print/building-a-basic-v4-printer-driver.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @barrygolden
* Microsoft Alias: **bagold**

Voidgear, 2019-11-02:
Microsoft is treating it's customers like trash, right? It's been almost 6 months.

frazeradam, 2019-11-22:
To save other people the trouble of wasting hours trying to figure out what the missing step is, it's that you need to install WDK: https://docs.microsoft.com/en-us/windows-hardware/drivers/other-wdk-downloads

2lazy2debug, 2020-12-02:
as an addendum to @frazeradam , once wdk is installed, it's going to prompt to install an extension for Visual Studio, do not forget to do that. I did, I looked stupid for a while.

closed 2021-01-27 by barrygolden