MicrosoftDocs/windows-driver-docs #1545: Missing Arg1=0x41201

jgyngell, 2019-05-10:
MEMORY_MANAGEMENT (1a)
    # Any other values for parameter 1 must be individually examined.
Arguments:
Arg1: 0000000000041201, The subtype of the bugcheck.
Arg2: ffffd28000087c98
Arg3: 0000001000000001
Arg4: ffffad033a39db30

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: de955911-dfbf-33b3-778a-767a5a578a0d
* Version Independent ID: 23fb5cbd-5f7d-f7ca-1622-5885020966c2
* Content: [Bug Check 0x1A MEMORY_MANAGEMENT - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x1a--memory-management)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0x1a--memory-management.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0x1a--memory-management.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2019-06-21:
@jgyngell - I have draft text prepared and being reviewed by the dev team for this parameter 1 value for bug check 0x1A.

DOMARS, 2019-07-01:
@jgyngell - The description for this bug check 0x1A parameter 1 is live. Thanks for your interest in improving the stop code documentation.

closed 2019-07-01 by DOMARS

agarwalg271, 2019-08-26:
Missing Arg1=0x5150b