MicrosoftDocs/windows-driver-docs #1557: Member not documented

BrianCatlin, 2019-05-19:
When will SyncType be SyncTypeCreateSection, and when will it be SyncTypeOther? What are the implications?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 20a64a62-c8b8-d3f7-942a-145b8d8907c9
* Version Independent ID: 3ecdda0f-ef9b-5bde-431e-81a7af013f64
* Content: [FLT_PARAMETERS for IRP_MJ_ACQUIRE_FOR_SECTION_SYNCHRONIZATION union - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ifs/flt-parameters-for-irp-mj-acquire-for-section-synchronization)
* Content Source: [windows-driver-docs-pr/ifs/flt-parameters-for-irp-mj-acquire-for-section-synchronization.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/ifs/flt-parameters-for-irp-mj-acquire-for-section-synchronization.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

lorihollasch, 2019-06-18:
@BrianCatlin , I've add descriptions for SyncType members - thanks for pointing out that they were missing. Implications depend on what miniport is doing here (registering a callback is optional).

closed 2019-06-18 by lorihollasch

BrianCatlin, 2019-06-18:
That is better, but I would point out that SyncTypeOther is used just before a section is being created, to allow a driver to acquire a file system resource lock that might cause a deadlock if it were done later, which is why it cannot be failed (because there are no constraints on what can be done at this time). Thus, an explanation of what sort of processing a filter/legacy fs driver can/should do is important (with some examples!), here. Also, knowing the sequence of events is very important to understanding. Treating a driver as a black block that just does what it is told at very specific times is a recipe for failure. The network doc team tried that, and it failed miserably (in the late 90's and early 2000's, network drivers and video drivers caused more crashes than any other driver type, according to Microsoft's Online Crash Analysis team), and it only abated after the docs were substantially improved. The driver writer MUST know what is happening under the covers in order to make decisions on how their driver should be implemented, as well as to debug and test it.