MicrosoftDocs/windows-driver-docs #1560: The reportdata API doesn't work correctly.

moolwalk, 2019-05-20:
The actual output is as below. What document says looks the output is a text file but it actually looks like a gz compressed file.
 
{
    "data": [
        {
            "ReportDatetime": "2019-05-20T08:00:00",
            "ReportStatus": "Success",
            "ReportLocation": "https://hardwareanalyticsprod.blob.core.windows.net/driver-reports-30038840/30038840_925_43334_201905200800.gz"
        }
    ],
    "errors": []
}

The bigger problem is if I try to get the gz file from the link specified in "ReportLocation", it just says ResourceNotFound. I doubt this API doesn't work at all.

Jesse Ahn
---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: d879ed43-026e-f068-5c18-8986273e184f
* Version Independent ID: 055a847f-30ef-72c2-784e-ee68713d628d
* Content: [Get report data - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/get-report-data)
* Content Source: [windows-driver-docs-pr/dashboard/get-report-data.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/dashboard/get-report-data.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **shganesh**

puffelyang, 2019-05-24:
Hey Jesse, 

This is normal. 

I had experience with this one, and the URL that you've got here will simply not work in browser, due to the concern of security. The return error 404 is just to avoid leaking any private information that might risk the security. 

So, this is by design of the Azure blob resource downloading. Since there's no security token in the URL, you'll need to supply authorization from the request header. 

However, there're still problem on my side as well,  when I'm trying to download the report. :( 

I've submit my problem to: https://github.com/microsoft/hpc-api/issues/1 and no one pickup yet. 

I hope this can help you to go further with your project, and even skip the problem I have to get the final success, and we can discuss about the issue I have then. 

Good luck and regards!
W.Y.

EliotSeattle, 2019-06-10:
Hi @moolwalk (Jesse).
It's not clear why you're having this issue, as this account was previously onboarded. The support teams asks that you please go ahead and open a support case from [https://support.microsoft.com/en-us/supportrequestform/83cdfd8d-c24a-fbe4-fb2a-3fead30613a9](https://support.microsoft.com/en-us/supportrequestform/83cdfd8d-c24a-fbe4-fb2a-3fead30613a9).

Thanks,

Eliot

closed 2019-06-10 by EliotSeattle