MicrosoftDocs/windows-driver-docs #1561: dual signature?

RonaldAi, 2019-05-20:
I checked driver signature on Windows 1903 and can only find SHA2 signature. Could you clarify which Windows versions use dual signatures and which build started using only SHA2 signatures?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: b5ca4e93-a34c-abbc-0ebc-9b5195378b20
* Version Independent ID: f2e19c4f-44cf-e0d4-2143-1e2129c582c9
* Content: [Driver Signing - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/driver-signing)
* Content Source: [windows-driver-docs-pr/install/driver-signing.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/driver-signing.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2019-06-15:
Hi @RonaldAi I just made commit 423a151ae with some extra info on this.  That commit should get synced over to this repo within 15 min, and the doc changes should be live in <30.  Thanks for the feedback!

closed 2019-06-15 by tedhudek

tedhudek, 2019-06-17:
Also see the new column on the right of https://docs.microsoft.com/en-us/windows-hardware/drivers/install/kernel-mode-code-signing-policy--windows-vista-and-later-#signing-requirements-by-version