MicrosoftDocs/windows-driver-docs #1562: Needs more detail

BrianCatlin, 2019-05-20:
Needs a link to where a hardware profile is described. What values in this tree would be accessed by a driver? (IoOpenDeviceRegistryKey allows this to be opened here, so there must be fields of interest to drivers)

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: da8ea7d6-362e-5e91-4ce8-f08e72547d48
* Version Independent ID: bd82463d-17de-d2cf-90b2-a7117acc49ea
* Content: [HKLM\SYSTEM\CurrentControlSet\HardwareProfiles Registry Tree - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/hklm-system-currentcontrolset-hardwareprofiles-registry-tree)
* Content Source: [windows-driver-docs-pr/install/hklm-system-currentcontrolset-hardwareprofiles-registry-tree.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/hklm-system-currentcontrolset-hardwareprofiles-registry-tree.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2019-09-27:
Hey @BrianCatlin apologies for the delay.  I double-checked with the feature team and the language on https://docs.microsoft.com/en-us/windows-hardware/drivers/install/hklm-system-currentcontrolset-hardwareprofiles-registry-tree is accurate, specifically that hw profiles are deprecated and shouldn't be used.  I added a note to https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdm/nf-wdm-ioopendeviceregistrykey to say the same, but it won't show up for another half hour or so.

closed 2019-09-27 by tedhudek

tedhudek, 2019-09-30:
In late breaking news, I got the following workaround from the team and have added this to the page:

Instead, store information globally using **PLUGPLAY_REGKEY_DEVICE** and **PLUGPLAY_REGKEY_DRIVER** without also using **PLUGPLAY_REGKEY_CURRENT_HWPROFILE**. For more info, see [**IoOpenDeviceRegistryKey**](https://docs.microsoft.com/windows-hardware/drivers/ddi/content/wdm/nf-wdm-ioopendeviceregistrykey).