MicrosoftDocs/windows-driver-docs #1565: Add hyperkbd to list of service options for forcing Bug Check Code

AlexSmith268, 2019-05-21:
For users that might need to force a bug check, crash dump, or BSOD on a Hyper-V based VM, you can add a DWORD of CrashOnCtrlScroll with a vaule of 0x01 to the Parameters subkey of the hyperkbd service.

Current Doc: https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/forcing-a-system-crash-from-the-keyboard

AlexSmith268, 2019-05-21:
If it helps, here is what you could add to to this doc to cover this...

Add the following to the beginning list:
Hyper-V keyboards

Add a forth item to the list that states the following:
4. With Hyper-V keyboards, you must enable the keyboard-initiated crash in the registry. In the registry key HKEY_LOCAL_MACHINE\System\CurrentControlSet\Services\hyperkbd\Parameters, create a value named CrashOnCtrlScroll, and set it equal to a REG_DWORD value of 0x01.

Add the following for the bullet list for alternate keys:
For Hyper-V keyboards:
HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Services\hyperkbd\crashdump

DOMARS, 2019-06-08:
@AlexSmith268 - Thanks for this sugestion. Will need to check with the devteam to confirm this is a supported method.

DOMARS, 2019-06-21:
@AlexSmith268 - Following up with the dev team on this.

DOMARS, 2019-07-01:
@AlexSmith268 - This content is now live. Thanks for pointing this out and filing this git issue.

closed 2019-07-01 by DOMARS