MicrosoftDocs/windows-driver-docs #1575: Is there a way to cancel a submission/product/commit?

puffelyang, 2019-05-23:
When testing, I noticed that for some unsuccessful submission,  it seems like the product will not show up in the portal. However, after a successful submission, the previously unsuccessful submission will also show up in the portal together with the successful one. Although, it seems like the commit request to specify the product id and submission id. It seems to be a bug to me, or a Cancel API for the actions before submission would be necessary. 

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 0ec4d9d1-95d0-aaf4-5bc2-1fd0ff3d6ed0
* Version Independent ID: a506d854-e239-58d1-342c-99622ee5249b
* Content: [Manage product submissions - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/manage-product-submissions)
* Content Source: [windows-driver-docs-pr/dashboard/manage-product-submissions.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/dashboard/manage-product-submissions.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @balapv
* Microsoft Alias: **balapv**

SillyKeith, 2019-05-23:
In the current design, the main starting "search" page is refreshed every 10 mins or so.   To view all results as they are created, you must use the top link named "Driver List Page (all)".   That page does take a bit to process and load if you have a lot of submissions, but the unsuccessful submission should show up in that ALL list.

EliotSeattle, 2019-05-23:
Added a note to the topic explaining this behavior. Should be live shortly.
Thanks, @puffelyang for the issue and  @SillyKeith for the response.
Eliot

closed 2019-05-23 by EliotSeattle