MicrosoftDocs/windows-driver-docs #1576: Many escape meanings reference "Windows SDK documentation" that cannot be found

ablaylock, 2019-05-23:
Many of the escape codes meaning column reference "the Windows SDK documentation" but when I search the documentation I cannot find any reference other than this page that discuss the escape code.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 38dfac08-3699-5396-f4aa-29a3a488d591
* Version Independent ID: 57cd2706-a965-7ab7-1d8d-118c70cb0a64
* Content: [Pscript-Supported Escapes - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/print/pscript-supported-escapes)
* Content Source: [windows-driver-docs-pr/print/pscript-supported-escapes.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/print/pscript-supported-escapes.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @barrygolden
* Microsoft Alias: **bagold**

barrygolden, 2021-01-28:
Adding an internal work item to update the Windows SDK links in the **Pscript-Supported Escapes** topic.

barrygolden, 2021-03-29:
Updated topic and added links to Windows SDK topics per user feedback.

closed 2021-03-29 by barrygolden