MicrosoftDocs/windows-driver-docs #1578: Any comments on the processing overhead ? "Assigning queues, and queuing additional DPCs requires additional processing overhead."

jagdishgopalakrishnan, 2019-05-25:
Interested to see if there are any additional time bounds that can be incurred based on the additional processing overhead needed for assigning queues and queueing additional DPCs.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 6b976194-de35-a0ee-7ea3-c7c1b1bf2e1f
* Version Independent ID: 081748f7-b9e4-2929-9619-da94ed518946
* Content: [RSS with a Single Hardware Receive Queue - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/network/rss-with-a-single-hardware-receive-queue#feedback)
* Content Source: [windows-driver-docs-pr/network/rss-with-a-single-hardware-receive-queue.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/network/rss-with-a-single-hardware-receive-queue.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @duncanmacmichael
* Microsoft Alias: **dumacmic**

duncanmacmichael, 2019-05-30:
Good question @jagdishgopalakrishnan, let me ask the NDIS team to see what they think.

jagdishgopalakrishnan, 2019-05-31:
Thanks, appreciate your help.

jtippet, 2019-05-31:
The docs say:

> Assigning queues, and queuing additional DPCs requires additional processing overhead. To achieve improved system performance, this overhead must be offset by better utilization of available CPUs.

This remark means that RSS doesn't magically make things faster.  Actually, RSS only adds overhead, since now you're paying for the overhead (ISRs, DPCs, function calls, etc) on multiple processors.

The purpose of RSS is to spread load across multiple CPUs, when 1 CPU isn't able to handle all the workload.  The docs are saying that, if 1 CPU is enough to handle your workload, then don't enable RSS; you'll only spend *more* cycles getting the same work done.  As with any scale-out situation in any technology, scaling isn't perfectly linear.  If you add more CPU cores, you increase the maximum amount of workload that can get done, but at the expense of slightly higher overhead.

jagdishgopalakrishnan, 2019-05-31:
Thanks, understood your comment about RSS purpose and the workload spread. 

I had a slightly different question about "Assigning queues, and queuing additional DPCs requires additional processing overhead" which I will try and rephrase.

It sounds like there are 3 different approaches according to my understanding:

1) RSS with single HW queue: Main ISR spawns off DPC, This reassigns buffers to CPU associated Receiver Queues and spawns yet another DPC on the CPU which has the receiver queue. 
2) RSS with NIC supported HW queues but with single interrupt: Here NIC already copies to CPU associated receiver queue. But since there is only one interrupt source, the main ISR needs to spawn a DPC on the CPU whose queue is non-empty.
3) RSS with NIC supported HW queues but with MSI directly associated with CPU. Here directly the CPU of choice gets interrupted and a corresponding DPC on that CPU.

Approach 3 seems like the most desirable in terms of performance and efficiency as this saves the most time in terms of "Assigning queues, and queuing additional DPCs requires additional processing overhead". So just wanted to understand if you had any ballpark time savings that one might end up in approach 3, getting specifically from queueing and spawning additional DPCs.

jtippet, 2019-06-03:
Ah, gotcha.  First, your taxonomy is correct: those are the 3 different ways a NIC driver can do RSS.  (You could add an "approach 0" to your taxonomy: a NIC that only provides the RSS hash alongside each packet, but doesn't support any queues or multiple interrupts.)  Your conclusion is also correct: approach 3 is the most desirable for the OS.  (And between the first two, approach 2 is more desirable than approach 1.)

The drawback of these more desirable approaches is that they demand more of the hardware.  Approach 3 requires managing a collection of interrupt vectors and hardware queues, so it requires quite a bit more silicon than approach 1.

It's difficult to give you a one-size-fits-all benchmark, since every workload is different.  It'd be far more meaningful for you to identify the workload that's meaningful to you, and have you run the benchmark yourself.  But to give you some vague rules-of-thumb:
* Approach 1 means you have serious scalability bottleneck on CPU0 (or whichever CPU dispatches the ISR).  This will limit the maximum throughput you can achieve to something like 100k packets/sec.
* Approaches 1 and 2 add extra latency in scheduling the cross-processor DPC.  This varies wildly: if the system is only moderately loaded, the cross-processor DPC might only incur a few microseconds of delay.  But if the system is more heavily loaded, then you can start seeing an extra few milliseconds of wasted time.  This isn't the end of the world, but remember that TCP is fairly sensitive to its RTT estimation: if TCP thinks the RTT goes up, it'll clamp down on your throughput.

I don't want to paint a gloomy picture: there are commercial NICs that use all three approaches, and having *any* RSS support is better than nothing.  But if you have choices, approach 3 gives you better latency and better scalability.  The top-tier server-grade NICs all use approach 3.

In any case though, the ultimate proof is in the measurement.  A NIC driver might, on paper, use the fanciest RSS approach, but still might have inferior performance because it was implemented carelessly.   At the end of the day, you can use theory to guide your measurement strategy, but the ultimate decision-maker has to be real measurements on your actual workloads.

If you fire up Windows Performance Toolkit (wprui.exe / wpa.exe), you can see exactly how much time is spent between an ISR and DPC, or between one DPC and another DPC.  For example, on one of my test systems, there's no contention for processor 0, so the DPC scheduling delay is a fairly minimal 30 microseconds.  But if multiple devices are trying to put DPCs on that processor, you'd see that number go way up, since one DPC gets queued behind another.  (This particular NIC uses approach 3, so there's no 2nd DPC on a different processor.)

[image]

jagdishgopalakrishnan, 2019-06-04:
Thank you very much for a very detailed response, 

For more specific questions on approach 3, I think its probably better if we can directly correspond with you (NDIS Driver Team) (jagdish.gopalakrishnan@intel.com). It would be great if we can internally facilitate a discussion.

Thanks in advance.

duncanmacmichael, 2019-06-04:
Great, thanks @jtippet for the great info. @jagdishgopalakrishnan I'll go ahead and close this issue since you are taking this offline with the NDIS team. Thanks!

closed 2019-06-04 by duncanmacmichael

jagdishgopalakrishnan, 2019-06-04:
Hello Duncan,

I know you closed this issue, but would appreciate contact info for direct correspondence (jagdish.gopalakrishnan@intel.com) with jtippet or other NDIS driver team members for further offline discussion on option 3.

duncanmacmichael, 2019-06-05:
Hi @jagdishgopalakrishnan, 

I'm not a member of the NDIS team, but I'll forward your request to them for further discussion here.

Thanks!

jagdishgopalakrishnan, 2019-06-05:
Thanks very much.