MicrosoftDocs/windows-driver-docs #1585: Links broken

kkoeman, 2019-05-29:
The links for the HID parser APIs are broken.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 9bf1b7f1-a59a-d9ef-7c8e-b994946afb32
* Version Independent ID: 3fcf10e3-c148-702e-9d9e-01bd87136d61
* Content: [Introduction to HID Concepts - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/hid/introduction-to-hid-concepts#feedback)
* Content Source: [windows-driver-docs-pr/hid/introduction-to-hid-concepts.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/hid/introduction-to-hid-concepts.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @duncanmacmichael
* Microsoft Alias: **dumacmic**

duncanmacmichael, 2019-05-31:
Hi @kkoeman, thanks for bringing this to my attention. I've confirmed that there is a bug in our reference system that is preventing the entire Hidpi.h header from rendering, so I've reached out to get that fixed. Stay tuned.

duncanmacmichael, 2019-06-12:
Still working on getting an answer for where the content for this header went; thanks for your patience.

duncanmacmichael, 2019-07-29:
Hi @kkoeman, thanks for your patience. We've resolved the internal bug that was preventing the Hidpi.h header's content from rendering, so all links are now live.

closed 2019-07-29 by duncanmacmichael