MicrosoftDocs/windows-driver-docs #1587: Is this still relevant to windows 10 1809 x64 and newer?

studleylee, 2019-05-29:
Is this still relevant to windows 10 1809 x64 and newer?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 3aeb487d-b749-4f6e-9ff5-71be68bf7a09
* Version Independent ID: 44418645-8c6e-7387-ba3e-02f478816961
* Content: [Installing an Unsigned Driver during Development and Test - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/installing-an-unsigned-driver-during-development-and-test#feedback)
* Content Source: [windows-driver-docs-pr/install/installing-an-unsigned-driver-during-development-and-test.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/installing-an-unsigned-driver-during-development-and-test.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2019-06-18:
Hi @studleylee, apologies for the delay.  Is there a specific aspect of this page that you are wondering about?

tedhudek, 2019-06-19:
The answer is yes, everything on the page is still relevant.

closed 2019-06-19 by tedhudek

studleylee, 2019-07-16:
Lame Answer and not true. Why don't you guys update your stuff?