MicrosoftDocs/windows-driver-docs #1591: LiveKernelReports should be documented/referenced.

charliearcuri, 2019-05-30:
LiveKernelReports are a form a memory dump that don't involve a crash.  They can be analyzed in a substantially similar way to understand why a subsystem (audio, Wi-Fi or USB) was reset by the Windows kernel or executive subsystem.  Please add links to "live dumps" which should (IMO) be called by the name of the folder they appear in by default: LiveKernelReports.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 691d8b50-7ca2-7f88-1583-75a89c9aa4c3
* Version Independent ID: 7a6284cd-ddc8-9d01-e377-546bfc6134c7
* Content: [Varieties of Kernel-Mode Dump Files - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/varieties-of-kernel-mode-dump-files#feedback)
* Content Source: [windows-driver-docs-pr/debugger/varieties-of-kernel-mode-dump-files.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/varieties-of-kernel-mode-dump-files.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2019-06-08:
@charliearcuri - Thanks for this suggestion. I can add it to the backlog, but if you are able to create a PR with what you think this should be, that would accelerate this work.

DOMARS, 2020-12-18:
@charliearcuri - I have created and internal tracking item and have engaged the dev team to break out and better describe the live dump bug checks, the folders that they use, etc. so closing this issue. We have a draft topic in review. Thanks for filing this issue and good luck in your work with stop codes and the debugger.

closed 2020-12-18 by DOMARS