MicrosoftDocs/windows-driver-docs #1601: need better explanations of IGNORE_SHORT_PACKETS and SHORT_PACKET_TERMINATE policies

leo-zspace, 2019-06-07:
See discussions here:
http://libusbk.sourceforge.net/UsbK3/usbk_pipe_management.html
https://community.osr.com/discussion/181538/winusb-and-transaction-termination

The evidence suggests that IGNORE_SHORT_PACKETS is applicable to [out] bulk endpoints (with reversed semantics) but w/o access to WinUSB source code it is impossible to tell for sure.

It would be nice if documentation can explain it in more details and cleaner language.

IGNORE_SHORT_PACKETS "Completes a read request when a short packet is received or a certain number of bytes are read. If the file size is unknown, the request is terminated at a short packet."

It would be good not to use "file size" terminology in the statement above. It is actually not "file" but rather "data transfer" size.

Thank you,
 

[Enter feedback here]

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 947f85b5-07bc-45eb-4bb9-75302b0566b6
* Version Independent ID: 73454115-5c1c-a8ea-f65d-0a40116cb58b
* Content: [WinUSB Functions for Pipe Policy Modification - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/usbcon/winusb-functions-for-pipe-policy-modification#feedback)
* Content Source: [windows-driver-docs-pr/usbcon/winusb-functions-for-pipe-policy-modification.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/usbcon/winusb-functions-for-pipe-policy-modification.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @duncanmacmichael
* Microsoft Alias: **dumacmic**

duncanmacmichael, 2019-06-08:
Hi @leo-zspace, thanks for the thoughtful feedback. I'll check into this with the USB team; stay tuned.

duncanmacmichael, 2019-07-09:
Hi @leo-zspace, just as an update I've been in contact with the USB team about this, but unfortunately they're heads down in development at the moment, so I won't be able to get an answer for you until next month. Sorry for the long wait. I'll keep you posted whenever I learn something.

tedhudek, 2021-02-25:
I've been unable to get additional info about this for you, Leo -- and apologies that it took so long to get back to you.  I hope that you got your problem worked out in the interim. Note that you can always click Edit on most pages on docs.microsoft.com if you have learned any additional tips that you would like to add to a documentation page.  Thanks again for reaching out, and best of luck!

closed 2021-02-25 by tedhudek