MicrosoftDocs/windows-driver-docs #1602: Minor misinformation

alfaunits, 2019-06-08:
"User-mode deadlocks arise when multiple threads of one application have blocked each others' access to the same resources"
User mode code can access global resources, e.g. memory mapping or event objects. These can be shared between applications.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: dcd7823f-e9bd-27e0-5490-61ce0112e3f4
* Version Independent ID: d3cf7b68-dff5-955a-3910-575d78fcac87
* Content: [Debugging a Deadlock - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/debugging-a-deadlock)
* Content Source: [windows-driver-docs-pr/debugger/debugging-a-deadlock.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/debugging-a-deadlock.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2019-06-08:
@alfaunits - Can you clarify what you think the improved wording should be? Is it stating what the resource could be - global resources such as memory mapping or event objects? Or is it that this can happen within a single applications use of global resources?

alfaunits, 2019-06-08:
"multiple threads of one application" is incorrect. It can be if multiple applications as well.
Just as in kernel mode.

DOMARS, 2019-06-08:
@alfaunits - Thanks for the clarification, I see what you are saying. How does this updated wording sound?

"User-mode deadlocks arise when multiple threads have blocked each others' access to the same resources"

alfaunits, 2019-06-09:
As it can happen in both cases (within threads of a single application and even between threads of different apps), my suggestion is something along the lines of:
"User-mode deadlocks arise when multiple threads, usually of a single application, have blocked each other's access to the same resource". Perhaps add "However, multiple threads of multiple applications can also block each other's access to a global/shared resource, e.g. a global Event, Semaphore."
(do check the spelling of other's vs. others', I think it is other's, because it is singular other)

DOMARS, 2019-06-10:
@alfaunits - Thanks for the help in better describing the causes of user mode deadlocks. The updates are live.

closed 2019-06-10 by DOMARS