MicrosoftDocs/windows-driver-docs #1604: Out of date

BrianCatlin, 2019-06-11:
Building drivers with the  BUILD tool is no longer supported, yet there is still a need for export drivers. How do we create an Export Driver using Visual Studio?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 6d33e947-6ef6-456f-784c-6635929151f9
* Version Independent ID: c784fe86-b359-9805-30a3-37ae986b1bef
* Content: [Creating Export Drivers - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/kernel/creating-export-drivers)
* Content Source: [windows-driver-docs-pr/kernel/creating-export-drivers.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/kernel/creating-export-drivers.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

MartinDrab, 2019-10-13:
> Building drivers with the BUILD tool is no longer supported, yet there is still a need for export drivers. How do we create an Export Driver using Visual Studio?

You may create export drivers as projects in Visual Studio 2017 (I did not migrate to 2019 yet but I doubt anything major related to this topic changed there). I used the steps below to create several working export drivers, so you may try your luck with them as well.
* create a new project from a template, such as **Empty WDM Driver**.
* in the projects settings, set the **General -> Configuration Type** to **Dynamic Library (.dll)**.
* in the project settings, set the **Linker -> Advanced -> No Entry Point** to **Yes (/NOENTRY)** (see below for more details),
* add a module definition file to the project, exporting `DllInitialize` and `DllFinalize` as private symbols (i.e. they are not included in the import library file (.lib)).

The export driver does not need to have an entry point since it is never called by the kernel.. The kernel finds the `DllInitialize` symbol and invokes it instead. Similarly, the `DllUnload` symbol is located and invoked just before driver unload. Both `DllInitialize` and `DllUnload` follow their definitions in the outdated document, i.e.
```
NTSTATUS DllInitialize(PUNICODE_STRING RegistryPath)
{
  . . .
}

NTSTATUS DllUnload(void)
{
  . . .
}
```
Your module definition file may look like this:
```
LIBRARY mydriver
EXPORTS
  DllInitialize PRIVATE
  DllUnload PRIVATE
  . . . other exported symbols . . .
```

And that should be all.

I do not think the runtime library linked to such an export driver gets properly initialized but that should not be a big deal since the most of its functions is implemented by NTOSKRNL and the only part that really initializes (as far as I noticed) is the security cookie.

Also, I experienced quite an interesting behavior when loading an export driver at boot-time. The driver the export one depends on actually loads (and its `DriverEntry` gets executed) even if the `DllInitialize` routine of the export driver fails. So, you "standard" driver should be prepared also for this case. I do not know whether demand-start drivers follow the same pattern.

Cymoki, 2019-10-14:
Hey Brian,
Can I get more information from you about this? I can see that the current documentation should be updated to include VS2019 changes but from what I can tell its still possible to build an export driver. 
Could you provide the version of VS and WDK your using? What errors your seeing? Any additional information would be useful here. 
Thanks,
Cymon

BrianCatlin, 2019-10-14:
Yes, it can be done with VS - with a lot of fooling around with compiler/linker options. It should be documented and **supported in VS as a project type**. The docs certainly shouldn't be referencing a tool that hasn't been supported in the WDK for 6 or 7 years

tedhudek, 2020-10-23:
I've made the additions to the docs with the info that I have for this - closing it out

closed 2021-02-02 by tedhudek